insane-adding-machines/unicore-mx #22: Last block format problem

kuldeepdhaka, 2016-09-13:
https://github.com/insane-adding-machines/unicore-mx/blob/master/lib/usbd/class/usbd_msc.c#L790
this cause ms->block_count to have the "Last Logical Block Address"

and https://github.com/insane-adding-machines/unicore-mx/blob/master/lib/usbd/class/usbd_msc.c#L345 send the "Last Logical Block Address" to host ("READ CAPACITY Command: 25h")

BUT, https://github.com/insane-adding-machines/unicore-mx/blob/master/lib/usbd/class/usbd_msc.c#L369
`for (i = 0; i < ms->block_count; i++) {` shouldn't that be `<= ms->block_count` ?

kuldeepdhaka, 2016-09-13:
related? https://github.com/libopencm3/libopencm3-examples/issues/112

kuldeepdhaka, 2016-09-13:
boo!

danielinux, 2016-09-13:
something, test IRCBot

kuldeepdhaka, 2017-01-29:
closed: not relevant anymore.

closed 2017-01-29 by kuldeepdhaka